Ford/AVData #3: IMU - gravity removed from linear accel?

kylelindgren, 2020-03-19:
To confirm, has gravity already been removed from the linear acceleration component of the IMU data in the rosbags?

ankitvora7, 2020-03-20:
Hi @kylelindgren yes it is the gravity compensated acceleration.

kylelindgren, 2020-03-20:
Excellent, thank you!

closed 2020-03-20 by kylelindgren

slamhuangxin, 2020-04-21:
is there the imu data which doesn't remove the gravity from the linear acceleration?

slamhuangxin, 2020-04-26:
Hello, @kylelindgren  Do you know or find why the linear acceleration is so large, especially the z-axis direction, some almost 6g? thanks

slamhuangxin, 2020-04-26:
Hello, @kylelindgren Do you know or find why the linear acceleration is so large, especially the z-axis direction, some almost 6m/s^2? thanks

Saixiaoma, 2020-04-26:
@slamhuangxin @kylelindgren ,hi, i meet the same question, if you get a right answer. I hope you can give me a reply. Thank you so much

kylelindgren, 2020-04-26:
I have not been working with this dataset recently and unfortunately am unable to answer this question.

huanhexiao, 2020-12-15:
@ankitvora7 @kylelindgren hi guys, do u mean i can simply add gravity 9.8m/s^2 to the z-axis, if i want a real world data?