Ford/AVData #8: download the files of datasets

slamhuangxin, 2020-04-06:
Hello sir,
your datasets are only the file---2017-08-04(1.6TB) in your website. can i download a part of it?  the 1.6T is so big for me that i can't downlod it. 
Thank you so much.

ankitvora7, 2020-04-06:
Hi @slamhuangxin, yes the dataset in total is 1.6TB but you can download whatever you want. If you hit the small arrow on the right on the '2017-08-04' tab, you'll see a list of vehicles within which we have 6 logs. If you're working with images, the download size will be pretty big. But if you're working with other sensor data then ust download the rosbags which are of the order of 10s of GBs.

[image]

slamhuangxin, 2020-04-07:
Thanks, @ankitvora7 ,   Thank you so much for give me reply and detial description. Yestarday, i didn't hit the small arrow and made so a low mistake.  I can download the data now.

slamhuangxin, 2020-04-08:
Sorry, sir. @ankitvora7 . i have another question about your datasets. i see there are only pictures from 7 cameras and rosbag in this website.   Are the GPS data, IMU Pose trajectory and SLAM corrected ground truth pose  included in the rosbag?   Will the another data like lidar be privated in the future? Nowadys, i want to use your datasets to test my multi-sensor fusion method for  vehicle localization.
Thank you so much.

slamhuangxin, 2020-04-08:
Sorry, sir. @ankitvora7 . i have another question about your datasets. i see there are only pictures from 7 cameras and rosbag in this website. Are the GPS data, IMU Pose trajectory and SLAM corrected ground truth pose included in the rosbag? do the rosbag only included the picture of front-left camera? Will the another data like lidar be privated in the future? Nowadys, i want to use your datasets to test my multi-sensor fusion method for vehicle localization.
Thank you so much.
[image]

ankitvora7, 2020-04-10:
hi @slamhuangxin, all the poses are included in the rosbag. The SLAM corrected ground truth pose is in the "/pose_grount_truth" topic. The rosbag only contains the front left camera images just for the purposes of viewing the data in RVIZ. For any processing on the images, you should download the high resolution versions separately. We will release more data in soon. This data set and apt to test your multi sensor fusion localization since it contains, imu, gps, lidar and camera data. It also contains our localization results in the "/pose_localized" topic so that you can compare your results with ours.

slamhuangxin, 2020-04-13:
Thanks for your detail introduction and reply. In this rosbag, all data has been synchronize as your paper says?  I hope that i can see  your more data soon too.

ankitvora7, 2020-04-14:
@slamhuangxin Yes the data is synchronized. We will be releasing more data soon. For now we are collecting valuable feedback from users and rectifying any shortcomings for our future release.

closed 2020-04-14 by ankitvora7