Ford/AVData #10: multi_lidar_convert failed by using the Sample-Data

junjtang, 2020-04-15:
Hi all,

I used the sample data using following command:
$ roslaunch ford_demo demo.launch map_dir:=/tmp/ford/Map calibration_dir:=/tmp/ford/Calibration-V2/
$ rosbag play /tmp/ford/Sample-Data.bag
$ roslaunch ford_demo multi_lidar_convert.launch

The first two commands executed successfully. But I go following error in the third command:

[ERROR] [1586980154.104065718]: Failed to load nodelet [/velodyne_yellow_convert] of type [velodyne_pointcloud/CloudNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class velodyne_pointcloud/CloudNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2
[ERROR] [1586980154.104222810]: The error before refreshing the cache was: According to the loaded plugin descriptions the class velodyne_pointcloud/CloudNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2

Any Idea to fix this?

ankitvora7, 2020-04-21:
@junjtang Can you try what is suggested regarding the velodyne packages on this reply ? https://github.com/Ford/AVData/issues/9#issuecomment-617079923

cicly, 2020-04-22:
@junjtang 
sudo apt remove ros-kinetic-velodyne*
cd ~/catkin_ws/src/ && git clone https://github.com/ros-drivers/velodyne.git
You are installing both the official, released version (ros-kinetic-velodyne) and cloning from source.

junjtang, 2020-04-22:
When I run the catkin_make, I got the error to compile velodyne

In file included from /home/awsgui/catkin_ws/src/velodyne/velodyne_driver/src/lib/input.cc:56:0:
/home/awsgui/catkin_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:58:18: fatal error: pcap.h: No such file or directory
compilation terminated.
velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/build.make:62: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o' failed
make[2]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o] Error 1
CMakeFiles/Makefile2:4598: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all' failed
make[1]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 46%] Linking CXX executable /home/awsgui/catkin_ws/devel/lib/velodyne_laserscan/velodyne_laserscan_node
[ 46%] Built target velodyne_laserscan_node
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

junjtang, 2020-04-22:
> @junjtang
> sudo apt remove ros-kinetic-velodyne*
> cd ~/catkin_ws/src/ && git clone https://github.com/ros-drivers/velodyne.git
> You are installing both the official, released version (ros-kinetic-velodyne) and cloning from source.

When I run the catkin_make, I got the error to compile velodyne

In file included from /home/awsgui/catkin_ws/src/velodyne/velodyne_driver/src/lib/input.cc:56:0:
/home/awsgui/catkin_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:58:18: fatal error: pcap.h: No such file or directory
compilation terminated.
velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/build.make:62: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o' failed
make[2]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o] Error 1
CMakeFiles/Makefile2:4598: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all' failed
make[1]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 46%] Linking CXX executable /home/awsgui/catkin_ws/devel/lib/velodyne_laserscan/velodyne_laserscan_node
[ 46%] Built target velodyne_laserscan_node
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

junjtang, 2020-04-22:
have to install dependency
apt-get install libpcap0.8-dev

junjtang, 2020-04-22:
It now works fine, but I got the error for the sample file:
For frame [velodyne]: Frame [velodyne] does not exist

Is there any specific rosbag file with the proper velodyne frame?

ankitvora7, 2020-04-24:
@junjtang were you able to fix this ? Can you tell which branch are you using from the velodyne repo and what kind of system configuration do you have ?

junjtang, 2020-04-27:
@ankitvora7  I still miss the frame. I use 
git clone https://github.com/ros-drivers/velodyne.git
to get the master branch. I installed the ROS Kinetic on Ubuntu 16.04. Anything else you want to know?

ankitvora7, 2020-04-28:
@junjtang I tried again with the latest branch of the velodyne repo and in the system same as your configuration. It works fine on our end. Can you run the following commands in separate terminals and share the terminal output ?
```
roslaunch ford_demo demo.launch map_dir:=/path/to/map/folder/ calibration_dir:=/path/to/calibration/folder/
```
```
rosbag play /path/to/your/bag/file/name.bag
```
```
roslaunch ford_demo multi_lidar_convert.launch
```
Also, are you getting this error on terminal or Rviz window? If it is Rviz, can you manually type in the name of the fixed frame as 'body'?

junjtang, 2020-04-29:
hi @ankitvora7 

$roslaunch ford_demo demo.launch map_dir:=/home/awsgui/ford/Map calibration_dir:=/home/awsgui/ford/Calibration-V2
... logging to /home/awsgui/.ros/log/e14410a0-89f9-11ea-9ed8-0aa63d088a3e/roslaunch-ip-xxxxx.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ip-xxxxx:38089/

SUMMARY
========

PARAMETERS
 * /point_cloud_map_loader/neighbor_dist: 128.0
 * /point_cloud_map_loader/pcd_topic: /pointcloud_map
 * /point_cloud_map_loader/pose_topic: /pose_ground_truth
 * /point_cloud_map_loader/publish_rate: 0.25
 * /reflectivity_map_loader/neighbor_dist: 128.0
 * /reflectivity_map_loader/pcd_topic: /reflectivity_map
 * /reflectivity_map_loader/pose_topic: /pose_ground_truth
 * /reflectivity_map_loader/publish_rate: 1
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    extrinsics_broadcaster (ford_demo/extrinsics_broadcast.sh)
    point_cloud_map_loader (map_loader/point_cloud_map_loader)
    reflectivity_map_loader (map_loader/reflectivity_map_loader)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [12474]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e14410a0-89f9-11ea-9ed8-0aa63d088a3e
process[rosout-1]: started with pid [12487]
started core service [/rosout]
process[reflectivity_map_loader-2]: started with pid [12500]
process[point_cloud_map_loader-3]: started with pid [12505]
process[extrinsics_broadcaster-4]: started with pid [12506]
process[rviz-5]: started with pid [12530]
[extrinsics_broadcaster-4] process has finished cleanly
log file: /home/awsgui/.ros/log/e14410a0-89f9-11ea-9ed8-0aa63d088a3e/extrinsics_broadcaster-4*.log

$ rosbag play /home/awsgui/ford/Sample-Data.bag
[ INFO] [1588151863.257081657]: Opening /home/awsgui/ford/Sample-Data.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
 [RUNNING]  Bag Time: 1501822150.298375   Duration: 27.267981 / 27.272558               40.45 
Done.

$ roslaunch ford_demo multi_lidar_convert.launch
... logging to /home/awsgui/.ros/log/e14410a0-89f9-11ea-9ed8-0aa63d088a3e/roslaunch-ip-xxxx.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ip-xxxx:38780/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /velodyne_blue_convert/calibration: /home/awsgui/catk...
 * /velodyne_blue_convert/max_range: 130.0
 * /velodyne_blue_convert/min_range: 3.0
 * /velodyne_green_convert/calibration: /home/awsgui/catk...
 * /velodyne_green_convert/max_range: 130.0
 * /velodyne_green_convert/min_range: 3.0
 * /velodyne_red_convert/calibration: /home/awsgui/catk...
 * /velodyne_red_convert/max_range: 130.0
 * /velodyne_red_convert/min_range: 3.0
 * /velodyne_yellow_convert/calibration: /home/awsgui/catk...
 * /velodyne_yellow_convert/max_range: 130.0
 * /velodyne_yellow_convert/min_range: 3.0

NODES
  /
    velodyne_blue_convert (nodelet/nodelet)
    velodyne_green_convert (nodelet/nodelet)
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_red_convert (nodelet/nodelet)
    velodyne_yellow_convert (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[velodyne_nodelet_manager-1]: started with pid [13082]
process[velodyne_red_convert-2]: started with pid [13083]
process[velodyne_yellow_convert-3]: started with pid [13084]
process[velodyne_blue_convert-4]: started with pid [13085]
process[velodyne_green_convert-5]: started with pid [13086]

junjtang, 2020-04-29:
> @junjtang I tried again with the latest branch of the velodyne repo and in the system same as your configuration. It works fine on our end. Can you run the following commands in separate terminals [...]

Where can I type the name of the fixed frame?
[image]

cicly, 2020-04-30:
set frame_id:

<launch>
  <arg name="calibration" default="$(find ford_demo)/params/lidarIntrinsics.yaml" />
  <arg name="manager" default="velodyne_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="3.0" />
  
  <arg name="lidar_red_frame" default="/lidar_red" />
  <arg name="lidar_yellow_frame" default="/lidar_yellow" />
  <arg name="lidar_blue_frame" default="/lidar_blue" />
  <arg name="lidar_green_frame" default="/lidar_green" />
  
  
  
  <node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" />

  <node pkg="nodelet" type="nodelet" name="velodyne_red_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    
    <remap from="velodyne_packets" to="lidar_red_scan"/>
    <remap from="velodyne_points" to="lidar_red_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_red_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_yellow_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    
    <remap from="velodyne_packets" to="lidar_yellow_scan"/>
    <remap from="velodyne_points" to="lidar_yellow_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_yellow_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_blue_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    <remap from="velodyne_packets" to="lidar_blue_scan"/>
    <remap from="velodyne_points" to="lidar_blue_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_blue_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_green_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    <remap from="velodyne_packets" to="lidar_green_scan"/>
    <remap from="velodyne_points" to="lidar_green_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_green_frame)"/>
  </node>
</launch>

cicly, 2020-04-30:
@junjtang set frame_id
`

<launch>
  <arg name="calibration" default="$(find ford_demo)/params/lidarIntrinsics.yaml" />
  <arg name="manager" default="velodyne_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="3.0" />
  
  <arg name="lidar_red_frame" default="/lidar_red" />
  <arg name="lidar_yellow_frame" default="/lidar_yellow" />
  <arg name="lidar_blue_frame" default="/lidar_blue" />
  <arg name="lidar_green_frame" default="/lidar_green" />
  
  
  
  <node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" />

  <node pkg="nodelet" type="nodelet" name="velodyne_red_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    
    <remap from="velodyne_packets" to="lidar_red_scan"/>
    <remap from="velodyne_points" to="lidar_red_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_red_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_yellow_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    
    <remap from="velodyne_packets" to="lidar_yellow_scan"/>
    <remap from="velodyne_points" to="lidar_yellow_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_yellow_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_blue_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    <remap from="velodyne_packets" to="lidar_blue_scan"/>
    <remap from="velodyne_points" to="lidar_blue_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_blue_frame)"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="velodyne_green_convert"
        args="load velodyne_pointcloud/CloudNodelet $(arg manager)">
    <param name="calibration" value="$(arg calibration)"/>
    <param name="max_range" value="$(arg max_range)"/>
    <param name="min_range" value="$(arg min_range)"/>
    <remap from="velodyne_packets" to="lidar_green_scan"/>
    <remap from="velodyne_points" to="lidar_green_pointcloud"/>
    <param name="target_frame" value="$(arg lidar_green_frame)"/>
  </node>
</launch>
`

junjtang, 2020-04-30:
> @junjtang set frame_id
> 
> ```
> <param name="calibration" value="$(arg calibration)"/>
> <param name="max_range" value="$(arg max_range)"/>
> <param name="min_range" value="$(arg min_range)"/>
> <remap from="velodyne_packets" to="lidar_red_scan"/>
> <remap from="velodyne_points" to="lidar_red_pointcloud"/>
> <param name="target_frame" value="/lidar_red"/>
> ```

@cicly which file is this? should I add a new parameter frame_id and set the "body" as value?

ankitvora7, 2020-05-06:
@junjtang were you able to fix this?

ankitvora7, 2020-05-06:
> > @junjtang were you able to fix this?
> 
> not yet, please explain me where to set the frame id: which config file and which parameter, thanks!

Do you get this issue only for blue or all lidars ? One debug step would be to publish a static transform between velodyne and lidar_blue frame using 
```
rosrun tf static_transform_publisher 0 0 0 0 0 0 velodyne lidar_blue 10
```
Remove the visualization for other lidars and see if you can visualize only blue lidar

cicly, 2020-05-07:
@junjtang 
[image]

cicly, 2020-05-07:
@junjtang set “multi_lidar_convert.launch”

[image]

erksch, 2020-06-03:
The LiDAR data is not showing up for me either, I set the target_frame but got the following error:
```
Failed to transform from frame [/lidar_red] to frame [map]
```

Also I get the following warnings with `roslaunch ford_demo multi_lidar_convert.launch`:
```
[ WARN] [1591172983.355566304]: Timings not supported for model 64E
[ WARN] [1591172983.355619289]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ WARN] [1591172983.471584976]: Timings not supported for model 64E
[ WARN] [1591172983.471721223]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ WARN] [1591172983.544529317]: Timings not supported for model 64E
[ WARN] [1591172983.544747158]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ WARN] [1591172983.626970934]: Timings not supported for model 64E
[ WARN] [1591172983.627039056]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
 
```

ankitvora7, 2020-06-06:
> NO TIMING OFFSETS CALCULATED

Hi @erksch We pushed two fixes to master. Can you try both the master branch (please fetch it first) and the 'velodyne-model-fix' and see if either of them work for you? Make sure you have the bag file playing and you also run the demo.launch file

ankitvora7, 2020-06-06:
> > @junjtang were you able to fix this?
> 
> not yet, please explain me where to set the frame id: which config file and which parameter, thanks!

@junjtang Were you able to fix this ? If not, please fetch the master branch and see if you still have issues. The velodyne package was updated as a result we needed to modify certain parameters.

ankitvora7, 2020-06-29:
Closing this due to inactivity. Please reopen if there are issues.

closed 2020-06-29 by ankitvora7

vniclas, 2020-09-22:
Hi @ankitvora7
I was running into the same issues as described above (CloudNodelet doesn't exist and NO TIMING OFFSETS CALCULATED). The first was fixed by replacing all occurrences of `CloudNodelet` with `TransformNodelet` after I realized that the velodyne package (pulled the master branch today) doesn't contain CloudNodelet anymore. The second issues was resolved by using your branch `velodyne-model-fix`.
On top of that, I also needed to remove all leading slashes ("/") in all fixed and target frames, e.g., `lidar_red` instead of `/lidar_red`. Using a slash resulted in crashing the nodelet: _Warning: Invalid argument "/lidar_red" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like ..._
I'm using melodic on ubuntu 18.04.