Ford/AVData #9: question about the time interval of IMU

slamhuangxin, 2020-04-15:
Hi, bro,
i get the gps and imu csv data from the same-data.bag by using your providing scripts---- bag _to_csv.py
However,   i find the time interval of IMU or GPS is a constant value.

For example, this the data of GPS in the csv format which i get.
        timestamp, latitude, longitude, altitude ,position_covariance
1501822123.041313792,  42.30597,-83.24483,156.02139,0.00000
1501822123.049295872,  42.30597,-83.24483,156.02137,0.00000
1501822123.051310080,  42.30597,-83.24483,156.02135,0.00000
1501822123.059303168,  42.30597,-83.24483,156.02133,0.00000
1501822123.061313024,  42.30597,-83.24483,156.02131,0.00000
1501822123.069293056,  42.30597,-83.24483,156.02129,0.00000
1501822123.071315968,  42.30597,-83.24483,156.02128,0.00000
.............................................................................
The time interval should be 0.005s because the frequency rate is 200Hz.
But the time interval of first and second timestamps is about  0.008s,  the time interval of second and third timestamps is about  0.002s, the average time interval is 0.005{ [0.008+0.002]/2 }. the situation does a loop in the later all lines. 
 I also try the another rosbag, the average time interval is also 0.005{ [0.003+0.007]/2 }.

Is the gps and IMU like that in fact ? 
Thank you so much. It is really very kind of you .I hope you can give me a reply.

cicly, 2020-04-21:
sudo apt remove ros-kinetic-velodyne*
RUN cd ~/catkin_ws/src/ && git clone https://github.com/ros-drivers/velodyne.git
You are installing both the official, released version (ros-kinetic-velodyne) and cloning from source.

ankitvora7, 2020-04-21:
@slamhuangxin What you are seeing is an artefact of a real robotics system. Even though the average frequency of data is 200hz, there are times when this is not met due to CPU load and similar other factors.

slamhuangxin, 2020-04-22:
ok. thank you so much. In addtion,is  the measuring data of linear accelaration in the  vehicle body frame coordinate system or in the imu frame coordinate system?

ankitvora7, 2020-04-28:
@slamhuangxin Vehicle frame coordinate system

closed 2020-05-06 by ankitvora7