Ford/AVData #12: Converting ground truth poses(location) into WGS84 frame

ZXia221, 2020-04-22:
Hi,

Thank you for your work!
Do you provide any code for converting the ground truth pose(location) from the local frame into WGS84 (GPS) coordinate frame?

agarwalsid, 2020-04-22:
@ZXia221, the origin is located at 42.294319, -83.223275 (Lat, Long). The map and poses are with respect to this origin. You can use this information to transform all poses into GPS coordinate frame.

ZXia221, 2020-04-23:
@agarwalsid, yes, you are right. 
From WGS84 to the local frame, I guess that the (lat, long) is first converted to (Northing, Easting) in the UTM frame. Using the origin, I can calculate the translation between UTM and the local frame. But, is there any rotation between these two coordinate frames? If not, which axis (x or y) in the local frame is aligned with the Northing?

ZXia221, 2020-04-23:
A follow-up question: how did you linearize the WGS84 frame? I tried to flatten the oblate spheroid into the UTM frame, then apply the translation on it to get local coordinates. But the difference between the resulting trajectory and the given trajectory (raw pose and GT pose) reaches several hundred meters.

ankitvora7, 2020-04-23:
@ZXia221 In terms of the rotation, the map frame is NED (North,East,Down) whereas the UTM is ENU (East, North,Up) so you'll have to apply a transformation there. 
In terms of the difference between raw pose and GT pose, you should expect some differences since the raw pose drifts. Can you do the same analysis with localized pose ? Also can you share a plot/image ?

ZXia221, 2020-04-23:
Sorry, I think my statement was unclear.
I converted the origin 42.294319, -83.223275 (Lat, Long) into UTM (Northing, Easting) to get the origin in UTM.
After this, in order to get the GPS trajectory in the local frame, all the GPS measurements (Lat, Long) are transformed into UTM (Northing, Easting) and subtracted by the origin in UTM (Northing, Easting).
The difference between this resulting trajectory (local GPS trajectory) and raw pose is large. 

When you linearize the WGS84 oblate spheroid, which global frame do you use? In other words, what is 'll2xy()' function in the figure below. Does ll2xy mean WGS84 to UTM?

[image]

ankitvora7, 2020-04-23:
@ZXia221 apologies for the confusion. If you look at section 7.5 of this document (https://portal.opengeospatial.org/files/16-011r4) it talks about converting angular displacements in WGS84 to linear displacements in X and Y. It uses the dlat and dlon (which is nothing but the difference between lat,lon of any point and the lat, lon at the origin point) and converts it to X and Y.

ZXia221, 2020-04-24:
Problem solved. 
Thanks a lot!

closed 2020-04-24 by ZXia221

juliajing72, 2020-08-11:
> Sorry, I think my statement was unclear.
> I converted the origin 42.294319, -83.223275 (Lat, Long) into UTM (Northing, Easting) to get the origin in UTM.
> After this, in order to get the GPS [...]

Hi, sorry, but I was wondering how you managed to solve the problem of reducing the difference between the converted GPS coordinates and the pose coordinates? 
thank you