Ford/AVData #13: About the linear accelaration

Saixiaoma, 2020-04-26:
Hi，dear authors,
I have read your paper and your websites about this AVData. I have the question about the linear accelartion. and I know it has removed the gravity and represents the oriention of body frame in the vehicle.
The imu.csv is transformed by your providing scripts. But the linear accelartion is so big like  this and i can't understand the reason.
[image]

I really hope you can give me a reply . 
Thank you so much

Saixiaoma, 2020-04-26:
[image]

ankitvora7, 2020-04-28:
Hi @Saixiaoma , can you tell us which log and time interval does this data snapshot belong to so that we can look on our end? Usually, you can expect spikes in imu data since it is subject to a lot of noise including the vibrations of the vehicle. You can remove this noise by using a low pass filter/moving-average filter or a KF

huanhexiao, 2020-11-21:
Hi guys, mine by contrast is so small,  will u please explain this for me?
[image]

closed 2020-11-24 by Saixiaoma