Ford/AVData #14: Direct access Rosbags on S3

wiegelmann, 2020-05-13:
Is it possible to put the Rosbags in a non-zipped form on S3, just the .bag files?

We have a bunch of analytics tools that can parse and read Rosbags directly on S3 w/o copying and converting. Makes sense if you process petabytes of bags per day, exploring the data and searching for edge cases.

Many thanks for your release of public AV data! That helps a lot.

ankitvora7, 2020-05-14:
Hi @wiegelmann, the rosbags were zipped so that it is easier/quicker for us to upload and users to download. Let us internally discuss and get back to you on the feasibility of uploading direct rosbags.

wiegelmann, 2023-02-04:
@ankitvora7 Any updates on your internal discussions?

Is the data released under MIT license, too?

closed 2023-05-30 by agarwalsid