Ford/AVData #17: Rosbag Timezone Information

kevinsoucy, 2020-06-25:
Hi, 

In general, how do you handle timezone information in the Rosbag file format?

Do you store timezone information anywhere in the Rosbag files or are all timestamps local time depending on the origin of the trip or even the current position of the car?

Thanks!
Kevin

ankitvora7, 2020-06-29:
@kevinsoucy The timestamps are epoch time in UTC time. E.g the first timestamp of the CENTER images of sample data is 1501822123109584 microseconds which is Friday, August 4, 2017 4:48:43.109 AM. Now, since the system clock drifts over time, this conversion is not very reliable (depending on what you're trying to do with it). The actual time when this data collection occurred was around 4:30 PM on August 4, 2017.

kevinsoucy, 2020-06-29:
Hi @ankitvora7 , Thanks for the response, I wanted to do enrich the data with a lookup of the weather conditions at the various lat/long's and timestamps. I think if the timestamps are +/- 30 minutes, it is precise enough for my use case at the moment.

closed 2020-07-01 by kevinsoucy