Ford/AVData #18: Online LiDAR scans in sensor frame or body frame?

troiwill, 2020-06-27:
Hello. I extracted LiDAR scans from the ROS bag with the help of **multi lidar convert.launch**. However, I am not sure which coordinate frame the LiDAR point clouds are in. Are the LiDAR point clouds in the respective LiDAR frames or in the body frame?

troiwill, 2020-06-28:
I found an answer to my question via the link (https://avdata.ford.com/data/default.aspx) under section **Coordinate Frames**. The section mentions that sensor data is recorded in the sensor frame and must be transformed into the body frame. So, for example, if I want to transform points from any LiDAR to world frame, I need to apply the following transform:

**point_in_world_frame**   =   **world_to_body_transform**   *   **body_to_lidar_transform**   *   **point_in_lidar_frame**

Is this correct?

ankitvora7, 2020-06-29:
@troiwill You are almost there! the transformations should be flipped -  

point_in_world_frame = body_to_world_transform * lidar_to_body_transform * point_in_lidar_frame

troiwill, 2020-06-29:
> point_in_world_frame = body_to_world_transform * lidar_to_body_transform * point_in_lidar_frame

@ankitvora7, thanks for the correction! 😄 Much appreciated.

closed 2020-06-29 by troiwill