Ford/AVData #20: IMU to GPS antenna lever arm

juliajing72, 2020-08-11:
Hi

Thank you for sharing the data and your work! I was wondering if you could also share the lever arm calibration between the GPS antenna and the IMU frame, or with respect to the vehicle body frame? 
thank you!

huanhexiao, 2020-11-21:
> Hi
> 
> Thank you for sharing the data and your work! I was wondering if you could also share the lever arm calibration between the GPS antenna and the IMU frame, or with respect to the vehicle body frame?
> thank you!

I think it is zero, since the imu data measurements and GPS BLH are both body frame

juliajing72, 2020-11-21:
> > Hi
> > Thank you for sharing the data and your work! I was wondering if you could also share the lever arm calibration between the GPS antenna and the IMU frame, or with respect to the vehicle body frame?
> > thank you!
> 
> I think it is zero, since the imu measurements and GPS BLH are both under body frame

Thanks, do you mean the GPS position outputs are already offset to the vehicle frame location, not the antenna location?

huanhexiao, 2020-11-23:
> Thanks, do you mean the GPS position outputs are already offset to the vehicle frame location, not the antenna location?

Yeah, i mean that. Generally speaking, GNSS/INS fusion should transform the result to IMU origin.

closed 2023-05-30 by agarwalsid