Ford/AVData #21: Velodyne data to pcd2 conversion

divyabhanukadaru, 2020-09-15:
Hello everyone,

Can we convert Ford LiDAR data which is available in **velodyne format** to **pcd2 format**? 
If yes, how can we convert that?

Thanks in advance

ankitvora7, 2020-09-15:
Hi @divyabhanukadaru you can install the velodyne package and then use the following command to convert the velodyne data to PCD2 format
`roslaunch ford_demo multi_lidar_convert.launch
`
Once you see the PCD2 in rviz, you can then use the pcl library to convert from pcd2 to .pcd files if you need.

JackRen88, 2021-08-12:
hello , @ankitvora7
I try to run this  `roslaunch`` ford_demo multi_lidar_convert.launch` for viewing the live lidar pointcloud,but It does not appear in rviz，there is a warnning that `[ WARN] [1628749506.660728638]: MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.velodyne_pointcloud.message_filter] rosconsole logger to DEBUG for more information.`
How should  I  solved it?

ArghyaChatterjee, 2021-09-12:
I am also having the same issue as @JackRen88 . @JackRen88 or @ankitvora7 do you have any idea why this is hapenning and how to solve this ?