Ford/AVData #22: Branch for latest version of velodyne_pointcloud compatible (ros-noetic)

Chris7462, 2020-09-20:
Hi all,

Since I'm using ros-noetic, I'm wondering is there a branch for latest version of velodyne_pointcloud compatible?

As far as I know, CloudNodelet has been removed from velodyne_pointcloud in velodyne-1.6.0 as transform node includes all features of cloud node. As a result, multi_lidar_convert.launch is not working and all nodes that load velodyne_pointcloud/CloudNodelet need to be changed.

I've checked velodyne-model-fix branch, but it does not have such changes.

Thanks in advance

ankitvora7, 2021-01-19:
Hi @Chris7462 - Apologies for the delayed response but I want to check if you were able to solve this issue ? If yes, please feel free to submit a PR

ankitvora7, 2021-01-19:
We have also pushed a noetic-devel branch that should work with the velodyne package

Chris7462, 2021-01-22:
Hi @ankitvora7, I tried to run `roslaunch ford_demo multi_lidar_convert.launch`, but still encounter issue as below. 

[image]

Chris7462, 2021-01-22:
Hi @ankitvora7, I think the value of "/lidar_red" should be changed to "lidar_red" in the multi_lidar_convert.launch file as below and the same for "/lidar_yellow", "/lidar_blue", and "/lidar_green".

[image]

ankitvora7, 2021-01-22:
Good catch @Chris7462 ! That's a typo there. Will correct it and push the update to the branch.

Chris7462, 2021-01-22:
Thank you @ankitvora7 ! This issue can be closed. Really appreciate it!

closed 2021-01-31 by Chris7462