Ford/AVData #23: question about IMU data

huanhexiao, 2020-11-21:
Hi authors, thanks for sharing ford avadata.
I extracted imu.csv from the .bag, whose linear accelaration is as following:
[image]

It seems too small and far away from gravity measurements 9.8m/s^2.
I've noticed some discussion about the transformation of imu.csv, but i cannot find where the script is.
Could anyone do me a favor please?

ankitvora7, 2020-12-22:
Hi @huanhexiao - The IMU data is processed data and as a result gravity is removed. You can add gravity vector after compensating for orientation.
To answer your second question, the origin of GPS data is the body frame.

huanhexiao, 2020-12-23:
@ankitvora7 Thanks very much, that's helpful.
Another question, I find that IMU sampling interval is not uniform. Here is some cases:

YanhaoZhang, 2022-02-18:
Hi @ankitvora7 I got the same question. May I know how is the gravity removed from the raw IMU data? Do you use some optimization approaches? If so, could you please provide some more details? Thank you.