Ford/AVData #24: [Question] integrate images and laser scans.

femust, 2020-12-22:
Hey, do you have some smart method to integrate lidar scans and images since as I saw the images are not synchronized explicitly with rosbag time steps, therefore some 'approximations' should be done, and I wonder whether it was your plan from the beginning in this dataset or you addressed somehow this issue? I am asking since I would like to avoid a method that I have a time step from rosbag and I look for the corresponding image-string that's "close".

Best

ankitvora7, 2020-12-22:
Hi @femust - You're right! The most obvious way to deal with this is to choose a camera image for each lidar scan (since lidar runs at a lower frequency than cameras). Also have a threshold for the difference between timestamps so that you're not using too old information.

femust, 2020-12-22:
Ok thanks for the info :)

femust, 2021-01-04:
[image]
Nah, I think that there is in general problem in rosbag with synchronization of lidar scans, since there is like a constant spatial offset between scans
[image]

ankitvora7, 2021-01-22:
@femust The lidar scans do not start at the exact same position and time. Moreover, these scans are not motion compensated. So each lidar will have different levels of effect of both the causes I mentioned.

closed 2023-05-30 by agarwalsid