Ford/AVData #26: Convert lidar csv files to pcds

Liumouliu, 2021-06-03:
Hi, 

   Thank you very much for the datasets. Can you please give detailed codes (write a script) to show how to convert lidar csv files to pcds?

   I have tried my best. But due to my poor ros knowledge, I failed.

   Thank you!

Best,
liu

Liumouliu, 2021-06-04:
Finally, I am able to do it. The details are listed below for the reference of other people who may want to do the same job.

1. roslaunch ford_demo multi_lidar_convert.launch 

By the time (2021/6/4), you need to revise the file multi_lidar_convert.launch following https://github.com/Ford/AVData/issues/10

If you want to skip the long replies, the ready-to-use file is attached (change the name to multi_lidar_convert.launch). [multi_lidar_convert.txt](https://github.com/Ford/AVData/files/6595131/multi_lidar_convert.txt)

2. Follow https://github.com/Ford/AVData/issues/21

install the pcl_ros library.

However, before install the library, plz change the line (https://answers.ros.org/question/331779/missing-ros-message-while-running-pointcloud-to-pcd/):

 sub_ = nh_.subscribe (cloud_topic_, 1,  &PointCloudToPCD::cloud_cb, this);

to 

sub_ = nh_.subscribe (cloud_topic_, 100,  &PointCloudToPCD::cloud_cb, this);

in file pointcloud_to_pcd.cpp

3. rosrun pcl_ros pointcloud_to_pcd put:=/lidar_blue_pointcloud

As I want to record the lidar blue pointclouds

4. slow play the rosbag, otherwise you will lost pcds

rosbag play ~/Sample-Data.bag -r 0.1

ankitvora7, 2021-06-05:
Some of these changes should be in the master branch!

ankitvora7, 2021-06-05:
Closing this issue

closed 2021-06-05 by ankitvora7

zl3466, 2023-03-26:
hello @Liumouliu and @ankitvora7 , thank you so much for the detailed instructions. May I ask what is the context of your successful retrieval of pcd files from the rosbag files (i.e. Ubuntu version, ros version)? I've been trying to reproduce your procedures on Ubuntu 20.04 with ros-noetic but am stuck with `roslaunch ford_demo multi_lidar_convert.launch
` with errors like:

`[ERROR] [1679794154.421029968]: Failed to load nodelet [/velodyne_blue_convert] of type [velodyne_pointcloud/TransformNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class velodyne_pointcloud/TransformNodelet with base class type nodelet::Nodelet does not exist.`

As far as I can learn this is caused by the incompatibility of the [velodyne package](https://github.com/ros-drivers/velodyne) with ros-noetic and Ubuntu 20. All I want is frame-by-frame pcd files that I can use for visualizations.

Also,  I am a bit confused why the csv files for lidars converted from bag_to_csv.py are full of rgb values but no x-y-z coordinates. Please let me know if you have any suggestions.